baldengineer/bit-preserve #5: Why no .pdf's included?

mrand, 2019-06-26:
Howdy,

Submitting here because I'm sure others have the same question.

Since it github's bandwidth anyway, I'm curious what the reason is for not including .pdf's along with the KiCad design?

I can somewhat understand not wanting pdf-only submissions (although I'd personally still like to see them), but by not including them even on KiCad designs, the utility of this repository is really limited to those that can successfully install KiCad.  There are a whole host of reasons that might not be the case:

1. Locked down computer (schools, or some work environments)
2. Bandwidth restrictions (especially a problem in poor countries, but even some people in developed countries have BW limits or data caps)
3. Incompatibility with KiCad.  There are various known issues
4. Already has a cad system installed, and either can't or don't want to install yet another (be it for security reasons, or resources, or who knows what).

I hope you'll reconsider - I think the goal here is noble and I'd like to see it have the greatest chance of success.

baldengineer, 2019-06-26:
The "no PDF" rule is meant for original scanned PDFs. In terms of project scope, I had in mind designs that predate eCAD generated PDFs. My decision was that we link to system-specific repositories that contain the original source material/schematics. 

Regarding KiCad generated PDF, I'm open to that.

mrand, 2019-06-27:
Ah, that's good to hear.  Might I suggest updating the "rules" readme since it seems to pretty explicit say only KiCad source files?

> Do not upload PDFs or images of schematics. It's okay to link back to a site hosting them (I guess). The only files to be hosted here are KiCad (aka editable) files.

DIYsciBorg, 2019-06-29:
Kicad can generate PDF files if/when needed, which would obviously be of better quality than old scans of original documents.
I agree that having pdf files along with the cad would be nice.
But the reasons mentioned as to one SHOULD provide pdf could just as easily apply to that format or any other (see more than others).. If anything, PDF files are less 'open' and would generally be of larger file size than the kicad file. 

Being composed primary of human readable text, the KiCad file would be much more searchable (both manually and machine) than PDF.
Demanding PDF output seems unnecessary, especially since anyone can do so on demand and include it back to the repo. Sure it would be NICE if the person making the scema could add the PDF but why make it a requirement?
As for the specific arguments:
1: locked down computers often can't view or download PDF or zip files either. People interested in retro hardware are not likely to be using such systems for their hobby anyway. I hate to be rude, but find another device/network where you can install the appropriate software. Note I agree here that PDF would be nice in such cases, but don't see the need to make it a rule.
2: bandwidth limits: where PDF is the WRONG option. More so when they want to download the whole repo. The zip file would be unnecessarily bloated due to additional PDF files which compress much less than the original KiCad file.
3: incompatibility: no idea what you mean here. If you mean changes due to version, kicads file upgrade features are very robust now, and clean upgrading is a priority on each new version.
4: different cad software: not relevant. KiCad is the software choice for very particular reasons. "Other" software generally implies commercial soft since KiCad is the only truely open/free option. Commercial software excludes more people than it includes. Even PDF "belongs" to Adobe. The point is to be  Free for all to install on multiple OS. Anyone with minimal cad soft skills can manage to load these files and do what they need to do. Some commercial soft is already starting to be able to import KiCad files one way as well.

Again, not saying PDFs would be useful or practical. Only that making it a rule seems unnecessary to me. Consider if the rule was "must include JPG so that we can embed in web pages, or must be viewable directly in github.?

Seems to me that more important things to work out include version numbering and contributor tracking. Did you copy the schematic layout page for page or did you optimize? How does one validate the accuracy and completeness of the schematic capture? Are board dimensions required? Etc

RobertBaruch, 2019-06-30:
PDFs and images will inevitably drift away from the KiCAD source of truth as changes are made either for more accuracy, better readability, or to correct errors.

It would be nice if there were an online generator of schematic -> image, but absent that, this repository should not be the place to solve that problem.

closed 2019-09-01 by baldengineer