baldengineer/bit-preserve #26: Symbols for specific applications of programmable generic parts

Cumbayah, 2019-07-12:
While working on capturing the Commodore 64 schematics, I came across a situation where I am putting symbols in the schematic that are a specific application of generic FPLAs and ROMs.

Examples include the PLA (CBM part number 906114-01) that is a specifically factory fused version of the Signetics 82S100, Fairchild 93459, MOS 7700 or MOS 8700 chips.
Similarly, the C64 ROMs are applications of generic 2364 and 2332 ROMs that are specifically mask programmed for application as BASIC, KERNAL and CHARSET ROMs.

In the PLA case, it's desirable to able to represent the specifics of the application in the symbol, to give meaningful pin names instead of just I0-15 and F0-8.

In the ROM case, whether chip select is active low or high is mask programmable so the generic symbol says CS/~CS - again, it's desirable to represent the specific application symbolically.

Yet, it is also desiable to retain the information about the generic parts that these specific applications are built from.

The way I propose to address these situations (and what I am currently doing in the symbol-libs contributed while working on my Commodore/C64/251469 WIP fork) is to add a symbol with the vendor specific application part number (e.g. 906114-01 for the C64 PLA) and then add aliases for the generic parts, prefixed with vendor, product and part. E.g. CBM_C64_PLA_82S100 for the Signetics FPLA application. The alias description can contain the additional information that this is an instance of such and such part in this specific application.

This way, one can still search on the generic part number in KiCad, but immediately see that it's a specific application and not a generic part that is matched. Also, the relation to the vendor part number and potential other generic applications is retained. Often there are no datasheets available for the specific application, in which case I still refer to the generic datasheets in the KiCad symbol definition.

Let me know if you see any issues with this approach or have directions for a better way to address this. Until then, I carry on like described in the above. :)

RobertBaruch, 2019-07-21:
I think that's reasonable.

baldengineer, 2019-07-25:
I think this approach makes sense. But I'll leave it open for a bit for others to comment.

closed 2019-09-01 by baldengineer