baldengineer/bit-preserve #50: Clarify license

jgrip, 2020-05-18:
Right now this repository doesn't have a license defined.

Should there be a global license for all of the schematics or would it be preferable to have a per-schematic license?

Personally I'm fine with pretty much any open source/HW license.

baldengineer, 2020-06-08:
I don't know what the right answer is. If I had thought about it more at the beginning, I would have picked a license like CC-BY-SA. But I don't know if that is fair to the work that others have done already.

I'm open to thoughts from others.

jgrip, 2020-06-09:
CC-BY-SA is fine with me.

@Cumbayah @RobertBaruch @georgjz @StormTrooper @nicolas-robin What are your thoughts?

StormTrooper, 2020-06-09:
I'm ok with CC-BY-SA

georgjz, 2020-06-09:
CC-BY-SA is fine by me

nicolas-robin, 2020-06-09:
I'm OK with CC-BY-SA

baldengineer, 2020-06-09:
Well, that was much easier than I expected. :) Thanks guys.

Cumbayah, 2020-06-09:
Sure, let's go with CC-BY-SA.

closed 2020-06-09 by baldengineer