baldengineer/bit-preserve #51: New Schematics

gbowne1, 2020-05-31:
1) are we adding Macintosh schematics for legacy Mac products, Lisa, Mac 128k, Mac 512k, Mac +, SE, SE/30, Classic, Classic II, LC's, II's (II, IIx, IIci, IIsi, IIfx, etc), Performas, Quadras, etc?  They are floating around the interwebs but no consistent repo with everything on it.

baldengineer, 2020-06-01:
I'm okay with extending the project to things in the 16/32-bit Mac project. But keep in mind, we don't want to become a PDF repository. 

If you're coming across places where they are hosted, the best option would be to create a directory structure and a README file with links to the PDFs.

gbowne1, 2020-06-01:
Ok, I am working on the a few like the Mac Plus and SE and SE/30.  The chips they use are pretty well documented just may not have .libs just figured if I get the basic chips and components in then someone with more KiCAD experience than me can connect the dots.  The Mac Plus schematics I found are fuzzy to read and theres two versions of the board.