baldengineer/bit-preserve #125: Apple II Schematic error

reifsnyderb, 2024-01-08:
On the Apple II Schematic, IO_STB off of F3 (74LS138), should be marked as follows:

IO Strobe - goes to all Peripheral Connectors (slots)
Goes low when accessing address range C800-CFFF

baldengineer, 2024-01-08:
FYI, It looks like the original submitter for the Apple II and II+ schematic either re-used the design or heavily based on the II+ on the II. 

I didn't realize that when I looked closer at the II+. So, I have put significant work into repairing that schematic. My plan was to "back-port" that one to the II design later.

We'll take feedback on either! But if you are doing a deep dive review, the II+ might be a better choice since it's a bit more active and has already had a number of corrections.

baldengineer, 2024-01-08:
> Sounds good. It's a process. :-)
>
> Also, I suspect the "INH" signal to inhibit the ROM from the expansion
> slots isn't tied to +5v...and should be. I haven't confirmed this yet, but
> have been unable to find the hold-up resistor.
>
> I really started with the II as I am exploring the possibility of an Apple
> II board using readily available electronics. i.e. Use a single SRAM chip
> instead of 24 DRAM chips.

For the most part, the only significant difference between the II and II+
(in terms of hardware) is the ROM. There were other minor changes, but the
vast majority of the chips are the same.

baldengineer, 2024-01-12:
Addressed in 2e13763 for the II+ schematic. The II schematic will get "fixed" when in a future effort.

commit 2e137632 (2024-01-12): fixed #125, F12 had the wrong address range listed and re-generated PDF
closed 2024-01-12 by baldengineer