baldengineer/bit-preserve #137: Apple IIe schematic mistake in clock divider area (UC1A/UC1B S109)

Tronix286, 2025-02-07:
Hi, 
There are mistake in clock divider area (UC1A/UC1B S109) in Apple/Apple IIe/Apple IIe (NTSC) - 050-0051-D/Apple IIe (NTSC) - 050-0051-D.pdf

[image]

Output Q pin (6) from first trigger must goes to J input pin (14) of second trigger
and output /Q pin (7) from first trigger must goes to /K input pin (13) of second trigger.

The correct wires diagram is drawn in the book [Jim Sather - Understanding the Apple IIe](https://archive.org/details/Understanding_the_Apple_IIe) but J and K second trigger mixed up:

[image]

I made a replica/clone of apple2e according to your scheme and it worked, only there were problems with the colors - they alternated correct/incorrect like so:

[image]

When i connected Q pin 6 to J pin 14 and /Q pin 7 to /K pin 13 its fixed color issues

[image]

rjhelms, 2025-02-07:
Good catch! I captured the schematic and it looks like my mistake is in the symbol - J and /K are in the right place in UC1B, but have the wrong pin numbers. J should be pin 13 and /K should be pin 14.

Should just be a matter of updating that symbol.

Tronix286, 2025-02-08:
Stop. Maybe iam wrong and your schematic is right. Today I traced the routing on the original Apple2e board using pictures on the Internet and it seems to match your diagram. It would be nice to check on a real apple2 with multimeter, I don't have one (that's why i'm creating a clone). Probably something is wrong with my board layout, I apologize for creating a fake issue.